STMicroelectronics/stm32-mw-usb-device #2: Include file versioning for every file

robcazzaro, 2020-09-24:
I frequently find example code and STM32 projects on the web using HAL files or middleware like USB. None of the files include version information, and it's almost impossible to find out what specific version a developer used. As a result, it's very hard to get things working since the changes from version to version are sometimes very significant.

I suggest creating a script or batch file in your production environment, so that every time you are ready for a release, you can include version information in the header of each file (where you now have the @file name.

Lack of versioning makes it impossible to reuse old code, and it's such a simple thing to implement

ALABSTM, 2021-01-08:
Duplicate of [STM32CubeF4#44](https://github.com/STMicroelectronics/STM32CubeF4/issues/44).

ALABSTM, 2021-01-08:
Hi @robcazzaro,

Thank you for having tackled the subject. Unfortunately, as answered previously, we do not plan to apply such a change.

With regards,

closed 2021-01-08 by ALABSTM

ApiumJacob, 2021-01-08:
> Hi @robcazzaro,
> 
> Thank you for having tackled the subject. Unfortunately, as answered previously, we do not plan to apply such a change.
> 
> With regards,

-1