STMicroelectronics/stm32-mw-usb-device #7: USBD_CtlSendData() is called with a pointer to stack allocated memory

jobasto, 2022-06-02:
### Problem:
In [Class/CDC/Src/usbd_cdc.c](https://github.com/STMicroelectronics/stm32_mw_usb_device/blob/555ce2b1f7960031f153c93ed4e1c0bb95f34c95/Class/CDC/Src/usbd_cdc.c), [Class/CDC_ECM/Src/usbd_cdc_ecm.c](https://github.com/STMicroelectronics/stm32_mw_usb_device/blob/555ce2b1f7960031f153c93ed4e1c0bb95f34c95/Class/CDC_ECM/Src/usbd_cdc_ecm.c), [Class/CCID/Src/usbd_ccid.c](https://github.com/STMicroelectronics/stm32_mw_usb_device/blob/555ce2b1f7960031f153c93ed4e1c0bb95f34c95/Class/CCID/Src/usbd_ccid.c), [Class/CCID/Src/usbd_ccid.c](https://github.com/STMicroelectronics/stm32_mw_usb_device/blob/555ce2b1f7960031f153c93ed4e1c0bb95f34c95/Class/CCID/Src/usbd_ccid.c) (and maybe others) `USBD_CtlSendData()` is called with pointer to stack allocated memory for its `pbuf` parameter.

### Example:
https://github.com/STMicroelectronics/stm32_mw_usb_device/blob/555ce2b1f7960031f153c93ed4e1c0bb95f34c95/Class/CDC/Src/usbd_cdc.c#L433
https://github.com/STMicroelectronics/stm32_mw_usb_device/blob/555ce2b1f7960031f153c93ed4e1c0bb95f34c95/Class/CDC/Src/usbd_cdc.c#L438
https://github.com/STMicroelectronics/stm32_mw_usb_device/blob/555ce2b1f7960031f153c93ed4e1c0bb95f34c95/Class/CDC/Src/usbd_cdc.c#L494

`USBD_CtlSendData()` does not read the content of `ifalt` immediately but only stores the pointer to it via
[USBD_LL_Transmit()](https://github.com/STMicroelectronics/STM32CubeF7/blob/e96f98542c6782d97bdd0c50833ee4b70130f3a7/Projects/STM32F769I_EVAL/Applications/USB_Device/CDC_Standalone/Src/usbd_conf.c#L547)->[HAL_PCD_EP_Transmit()](https://github.com/STMicroelectronics/stm32f7xx_hal_driver/blob/738d0ab28f9631b4fc4e865515c1867efe17d813/Src/stm32f7xx_hal_pcd.c#L1735). Reading only takes place when the [transmit fifo empty interrupt](https://github.com/STMicroelectronics/stm32f7xx_hal_driver/blob/738d0ab28f9631b4fc4e865515c1867efe17d813/Src/stm32f7xx_hal_pcd.c#L2000) fires. At that moment `USBD_CDC_Setup()` may already have returned and ifalt may have gone out of scope.

### Solution:
Declare `ifalt` as `const static`. Examine the code for other locations where stack allocated memory is passed to `USBD_CtlSendData()`. Modify `USBD_CtlSendData()`'s signature from
```
USBD_StatusTypeDef USBD_CtlSendData(USBD_HandleTypeDef *pdev,
                                    uint8_t *pbuf, uint32_t len)
```
to
```
USBD_StatusTypeDef USBD_CtlSendData(USBD_HandleTypeDef *pdev,
                                    const uint8_t *pbuf, uint32_t len)
```
to be able to pass a `const` pointer without cast.

billybednar, 2022-06-29:
The same applies to `status_info`, which is declared on the line after `ifault`.

ALABSTM, 2024-02-13:
Hi @jobasto,

First of all, please excuse this delayed reply. Many thanks for this analysis. Your point looks relevant after following your reasoning. May I ask if you could reproduce the issue you described before digging any further?

With regards,

ALABSTM, 2024-07-15:
Hi @jobasto,

I hope you are doing well. Please allow me to close this issue as no activity since a while. Thank you for your comprehension.

With regards,

closed 2024-07-15 by ALABSTM

jobasto, 2024-07-15:
Hi @ALABSTM,

I'm wondering why this issue is closed because it is clearly a bug. My first report should suffice to show that. Unfortunately, I'm not working on USB anymore so I'm not able to support you in fixing this bug.

Best regards,