STMicroelectronics/stm32-mw-usb-device #11: Does it support mouse and keyboard?

jianwa1, 2022-08-12:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus).

**Describe the set-up**
 * The board (either ST RPN reference or your custom board).
 * IDE or at least the compiler and its version.

**Additional context**
If you have a first analysis or a patch proposal, thank you to share your proposal.

**Screenshots**
If applicable, add screenshots to help explain your problem.

RJMSTM, 2022-11-09:
Hi @jianwa1 

First, thank you for your contribution, second could you please clarify your question in order to allow a better analysis of the problem.

With regards,

ALABSTM, 2023-05-15:
Hi @jianwa1,

Please allow me to close this issue as no reply to our question has been received since a while. Thank you for your comprehension.

With regards,