STMicroelectronics/stm32-mw-usb-device #12: Code is not clean!

sillfsxa, 2022-10-16:
If you try to find out which file uses which objects from other files (for example do it by drawing arrows between files based on the including of their header files), you'll find out that everything in this library needs access to conf.c, conf.h and def.h.
OK, then put all these things in just one file (for example with the name of conf.h) and include that in other files. Why there should be so much distributed objects that seems to be of the same nature.

On the other hand, if you draw the relationship between files, ioreq.c uses core.h and core.c uses ioreq.h!!!! this is a circle and it's bad!
It seems you are using something in core that does not belong to it and there are some objects in ioreq that belong to the core!

These were two issues that I found in the library and wanted to inform you about.

ALABSTM, 2023-06-12:
ST Internal Reference: 154588

ALABSTM, 2024-02-13:
Hi @sillfsxa,

I hope you are fine. Our development teams agree with your point. However, such an update would yield a backward-compatibility break. Hence, choice has been made to keep things the way they are.

Please allow me to close this issue. Thank you for your comprehension and thank you again for having reported.

With regards,