STMicroelectronics/stm32-mw-usb-device #14: CCID class

rayban9083, 2023-07-24:
I have seen that this middleware support for the CCID class

Unfortunately, there is no documentation or example implementation. 

Is there something you can share? 

Thanks in advance.

TOUNSTM, 2023-08-02:
Hello @rayban9083,

Thank you for this report. We will get back to you as soon as we analyze it further. This may take some time. Thank you for your comprehension.

With Regards,

light-tech, 2024-01-07:
For now the Cube IDE have not implemented the UI (code generation) for CCID class yet but I managed to imitate the code generation for the other USB classes to get it to work (see the `F411CEU6_USB_CCID` in https://github.com/light-tech/LearnSTM32). Of course, without official support, the code will get cleaned up when regenerating code again but I can always rollback undesired changes with Git.

There is also https://github.com/canokeys/canokey-stm32 that comes with an .ioc file but built with CMake and not Cube IDE. (Maybe the author used the IDE for 1 time code generation.)

ALABSTM, 2024-02-13:
Hi @rayban9083,

Please excuse this delayed reply. Please have a look at this [post](https://github.com/STMicroelectronics/STM32Cube_MCU_Overall_Offer/discussions/18). I hope this answers your question. Do not hesitate otherwise.

With regards,