STMicroelectronics/stm32-mw-usb-device #15: Support multiple CustomHID interfaces with CompositeBuilder

XelaRellum, 2023-09-25:
Right now there is one report descriptor size for all CustomHID interfaces, defined with `USBD_CUSTOM_HID_REPORT_DESC_SIZE`.

To adress this, the structure `USBD_CUSTOM_HID_ItfTypeDef` should contain a field with the actual report descriptor size.

TOUNSTM, 2023-10-03:
Hello @XelaRellum,

Thank you for this report. We will get back to you as soon as we analyze it further. This may take some time. Thank you for your comprehension.

With Regards,

ALABSTM, 2024-02-13:
Hi @XelaRellum,

Please excuse this delayed reply. 

Regarding the size of the report descriptor, I see this line in the `USBD_CUSTOM_HID_Setup()` function.

https://github.com/STMicroelectronics/stm32_mw_usb_device/blob/7b5e6886d2f11ad15d2b46364e8a96984358f639/Class/CustomHID/Src/usbd_customhid.c#L409

Structure member `wLenght` is defined in the structure below.

https://github.com/STMicroelectronics/stm32_mw_usb_device/blob/7b5e6886d2f11ad15d2b46364e8a96984358f639/Core/Inc/usbd_def.h#L197-L204

Maybe the report size could be defined there, in the frame of the setup request? Doing so, the size would be **any value** chosen at application level, and not necessarily the one defined by `USBD_CUSTOM_HID_REPORT_DESC_SIZE`.

With regards,

ALABSTM, 2024-02-13:
ST Internal Reference: 173362

ALABSTM, 2024-02-13:
Hi again,

Proposal forwarded to our development teams.

With regards,

ALABSTM, 2025-04-16:
Hi @XelaRellum,

You request has finally been considered in version **2.11.4** of the library, which has just been released. Below are some code snippets:

https://github.com/STMicroelectronics/stm32-mw-usb-device/blob/2a0a3521ac4d84e6e494d37bed615e2d36c373f5/Class/CompositeBuilder/Src/usbd_composite_builder.c#L1396-L1400

https://github.com/STMicroelectronics/stm32-mw-usb-device/blob/2a0a3521ac4d84e6e494d37bed615e2d36c373f5/Class/CustomHID/Inc/usbd_customhid.h#L106-L108

Thank you for your patience and thank you again for your proposal.

With regards,

closed 2025-04-16 by ALABSTM