STMicroelectronics/stm32-mw-usb-device #18: Missing code in usbd_cdc_if.c needed to build with USB Composite Builder.

JacobChrist, 2024-06-29:
**Describe the set-up**
 * The board (either ST RPN reference or your custom board).
Nucelo STM32H503
 * IDE or at least the compiler and its version.
STM32CubeIDE 15.1

**Describe the bug**
A clear and concise description of what the bug is.
The following functions in usb_cdc_if.c needed to be modified to get them to work with the composite builder:

``` c++
static int8_t TEMPLATE_Init(void)
{
#ifdef USE_USBD_COMPOSITE
    USBD_CDC_SetTxBuffer(&hUsbDeviceFS, UserTxBufferFS, 0, CDC_CLASS_ID);
#elif
    USBD_CDC_SetTxBuffer(&hUsbDeviceFS, UserTxBufferFS, 0);
#endif
    USBD_CDC_SetRxBuffer(&hUsbDeviceFS, UserRxBufferFS);
    return (0);
}

uint8_t TEMPLATE_Transmit(uint8_t* Buf, uint16_t Len)
{
  uint8_t result = USBD_OK;
  USBD_CDC_HandleTypeDef *hcdc = (USBD_CDC_HandleTypeDef*)hUsbDeviceFS.pClassData;
  if (hcdc->TxState != 0){
    return
             USBD_BUSY;
  }
#ifdef USE_USBD_COMPOSITE
    USBD_CDC_SetTxBuffer(&hUsbDeviceFS, Buf, Len, CDC_CLASS_ID);
    result = USBD_CDC_TransmitPacket(&hUsbDeviceFS, CDC_CLASS_ID);
#elif
    USBD_CDC_SetTxBuffer(&hUsbDeviceFS, Buf, Len);
    result = USBD_CDC_TransmitPacket(&hUsbDeviceFS);
#endif
  return result;
}
```

**How To Reproduce**
1. Indicate the global behavior of your application project.
Poorly since ST libraries are not documented well.

3. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...).
usbd_cdc_if.c

4. The use case that generates the problem.
Build a HID + CDC composite device using the Composite Builder.

5. How we can reproduce the problem.
Build a HID + CDC composite device using the Composite Builder and try to compile.

**Additional context**
I'm not sure if this fix alone is needed as I have not fully tested everything, but this minimally was needed just to compile.

**Screenshots**
If applicable, add screenshots to help explain your problem.

ALABSTM, 2024-07-08:
Hi @JacobChrist,

Thank you for this report. Two points I would like to draw your attention two, if I may:
* You mention a NUCLEO-H503RB as hardware setup. However, the STM32CubeH5 does not natively support ST's USB Device library, rather Azure [USBX](https://github.com/STMicroelectronics/STM32CubeH5/tree/main/Middlewares/ST/usbx) one. Hence, in case the issue is related to the **USBX** library, you can repost your issue [here](https://github.com/STMicroelectronics/stm32_mw_usbx).
* As the term `template` in the `usbd_cdc_if_template.c` filename suggests, the functions provided are just templates **to be customized** by the user according to their application's need, as clearly mentioned by the comment in the code snippet below.

https://github.com/STMicroelectronics/stm32_mw_usb_device/blob/0b06460a43c30670a764cb7e58c5a7918e789f6e/Class/CDC/Src/usbd_cdc_if_template.c#L97-L103

Now, you can find an example of a CDC standalone application on STM32**U5** series using ST's USB device library [here](https://github.com/STMicroelectronics/stm32u5-classic-coremw-apps/tree/main/Projects/NUCLEO-U575ZI-Q/Applications/USB_Device/CDC_Standalone). Although there is no `CDC_Itf_Transmit()` function, you can still have a look at how both `CDC_Itf_Init()` and `CDC_Itf_Receive()` functions have been implemented.

I hope this clarifies the question. Otherwise, I may have missed your point. Please do not hesitate to provide me with more clarifications in that case.

With regards,

ALABSTM, 2024-07-22:
Hi @JacobChrist,

I hope you are fine. I hope also my previous comment clarified the usage of the **template** files and they are intended to be customized by the user according to their application's needs. Please allow me to close this issue.

With regards,

closed 2024-07-22 by ALABSTM

JacobChrist, 2024-07-23:
Hmm, I think I got some code in the TEMPLATE from some place else that had only this:

``` c++
 static int8_t TEMPLATE_Init(void)
 {
     USBD_CDC_SetTxBuffer(&hUsbDeviceFS, UserTxBufferFS, 0);
     USBD_CDC_SetRxBuffer(&hUsbDeviceFS, UserRxBufferFS);
     return (0);
 }

uint8_t TEMPLATE_Transmit(uint8_t* Buf, uint16_t Len)
{
  uint8_t result = USBD_OK;
  USBD_CDC_HandleTypeDef *hcdc = (USBD_CDC_HandleTypeDef*)hUsbDeviceFS.pClassData;
  if (hcdc->TxState != 0){
    return
             USBD_BUSY;
  }
  USBD_CDC_SetTxBuffer(&hUsbDeviceFS, Buf, Len);
  result = USBD_CDC_TransmitPacket(&hUsbDeviceFS);
  return result;
}
```

None the less, the code I suggested would be helpful to new users of the library.

Open source software is easier to use than this supposedly helpful software.  Please make it easier.