STMicroelectronics/stm32-mw-usb-device #19: Will AUDIO class support UAC2?

xu-yingjing, 2024-07-04:
Hello,

I'm using this library on STM32H7RS's project to implement a USB sound device. The STM32H7RS has USB_OTG_HS ip and internal HS Phy, so it can easily achieve USB high speed. However, this library implements USB AUDIO Class based on UAC1.0 which does not support USB High Speed.

ALABSTM, 2024-07-08:
Hi @xu-yingjing,

Your question has been forwarded to our development teams. I will get back to you as soon as I have their answer.

With regards,

ALABSTM, 2024-07-08:
ST Internal Reference: 185997

ALABSTM, 2024-07-15:
Hi @xu-yingjing,

According to our development teams, the so called _classic core_ USB library **only** supports the UAC 1.0. In case you need UAC 2.0, please **move to** the **USBX** library.

You can find the library in [this](https://github.com/STMicroelectronics/stm32_mw_usbx) repository. STM32**H7RS** applications can be found [here](https://github.com/STMicroelectronics/x-cube-azrtos-h7rs).

With regards,

closed 2024-07-15 by ALABSTM