STMicroelectronics/stm32-mw-usb-device #20: SWV printf debugging dies when USB device connects

JacobChrist, 2024-09-17:
**Describe the set-up**
 * Custom board using a STM32H503CB.
 * Using this library because CubeMX doesn't support generating files for the STM32H503CB.
 * We are using printf debugging over SWV
 * When the STM32 USB device enumerates the SWV printf debugging breaks.
 * STM32CubeIDE Version: 1.15.1, STM32CubeMX version 6.11.0

**Additional context**
SWV printf debugging works up until the point that the STM32 enumerates.  I've tried to find the exact piece of code that is causing it to break by using printf debugging but because there is a delay from when the data is received and from when it is sent I don't think I'm finding where its breaking.

I was thinking that there was something in the USB-CDC class that was taking over printf but I can't find anything that might be doing this (and there is nothing coming out the COM port).

Please help

ALABSTM, 2024-09-23:
Hi @JacobChrist,

Have you had a look at the [stm32h5-classic-coremw-apps](https://github.com/STMicroelectronics/stm32h5-classic-coremw-apps) repository? There, you will find an [HID Standalone](https://github.com/STMicroelectronics/stm32h5-classic-coremw-apps/tree/main/Projects/NUCLEO-H563ZI/Applications/USB_Device/HID_Standalone) application based on ST's core USB Device library, that could be helpful.

Please let me know you could find a clue by browsing into the application files.

With regards,

ALABSTM, 2024-10-11:
Hi @JacobChrist,

Please allow me to close this thread as no feedback since a while. Please do not hesitate to log a new issue should your problem still persist. Thank you for your comprehension.

With regards,

closed 2024-10-11 by ALABSTM

JacobChrist, 2024-10-11:
And yet the problem persists....