STMicroelectronics/stm32-mw-usb-device #23: DFU Leave does not return dfuMANIFEST status

by-gnome, 2025-05-13:
**Describe the set-up**
STM32L4xx, STM32 USB Device Library V2.11.3

**Describe the bug**
In the current implementation of the DFU class, the DFU_Leave routine disconnects the USB device by calling USBD_Stop(). As a result, the device does not respond to the DFU_GETSTATUS request.
If USBD_DFU_VENDOR_EXIT_ENABLED is defined, it is possible to leave DFU mode asynchronously, but the USB device must not be disconnected.

**Possible solution**
```
--- a/Middlewares/ST/STM32_USB_Device_Library/Class/DFU/Src/usbd_dfu.c
+++ b/Middlewares/ST/STM32_USB_Device_Library/Class/DFU/Src/usbd_dfu.c
@@ -1337,13 +1337,13 @@ static void DFU_Leave(USBD_HandleTypeDef *pdev)
     hdfu->dev_status[3] = 0U;
     hdfu->dev_status[4] = hdfu->dev_state;
 
-    /* Disconnect the USB device */
-    (void)USBD_Stop(pdev);
-
 #if (USBD_DFU_VENDOR_EXIT_ENABLED == 1U)
     /* Jump should be ensured by user application */
     DfuInterface->LeaveDFU(hdfu->data_ptr);
 #else
+    /* Disconnect the USB device */
+    (void)USBD_Stop(pdev);
+
     /* Generate system reset to allow jumping to the user code */
     NVIC_SystemReset();
 #endif /* USBD_DFU_VENDOR_EXIT_ENABLED */
```

ALABSTM, 2025-05-16:
Hi @by-gnome,

Thank you for this report. Your analysis and fix proposal have been forwarded to our development teams. I will keep you informed.

With regards,

ALABSTM, 2025-05-16:
ST Internal Reference: 209913